UBC-Rocket/WB-DAQ-Firmware #3: Dynamic Pressure Regulation

misprit7, 2021-07-04:
Implement dynamic pressure regulations system in firmware. I suspect this will end up being quite complicated, will need to clarify requirements before we get started on this.

davinbirdi, 2021-12-03:
As of December 2, 2021 we've verified the high-side switches can actuate the Marotta Valve and have firmware implementation of direct enable of the high-side switches.

davinbirdi, 2022-01-09:
As of December 16th, 2021 we've verified ADC readings directly off the main chip using a potentiometer breadboard setup. Next step is to validate pressure transducers

davinbirdi, 2022-02-23:
Working and Tested P-Controller for DPR. Validated keeping flow rate constant using SQUIRTLE Test Stand.

Better Experimental Procedure and Recording are the next steps, but we have a validated prototype which is a huge step.