UBC-Rocket/WB-DAQ-Firmware #12: Duty Cycle of Valve

davinbirdi, 2022-01-16:
Be able to turn the valve from open to close at a set duty cycle

davinbirdi, 2022-01-17:
Thinking of using the Timer PWM Module (TPM) on the chip to do the duty cycle for the valve, @misprit7 let me know if you think this is a good idea or if I should write it fully contained within a RTOS Task

misprit7, 2022-01-18:
I think delays are actually better here given the low switching frequency of the valve. Presumably we'll have our control algorithm running at a relatively fast interval given there's essentially no limit on how fast we can read the pressure transducers, so I don't think it really makes sense to use PWM signals. The reason to do so is to give an intermediary value between loops of the control algorithm, but given the algorithm will be running faster than the valve can open and close anyways I don't really think there's a point. It's definitely something to consider though, especially if the valve turns out to have a much higher switching rate than we thought or it's possible to half open it with a pwm (unlikely) or something though.

davinbirdi, 2022-02-14:
Update: just using RTOS Delays we get some jitter in the period of PWM signal. This can vary up to 40ms for our 200ms period.

Need to debug and verify if there is some integer truncation occuring during the duty cycle calculation.

davinbirdi, 2022-02-17:
Still getting some Jitter, especially so when I was having the Actuator task running at the same period as the Control task.
This can vary quite a bit at lower periods.

davinbirdi, 2022-02-23:
After Feb22 DPR Testing: I noticed from the sound of the valve opening and closing that there was a distinct beat when running with the debug session open. When disconnected from the debugger, the chip runs without the debug configuration active skipping `printfs` and it sounds very steady without any pauses. 

I speculate the jitter is due to the printfs running in the control task.

This also doesn't make sense as the Control Task and the Actuator Task are separate and FreeRTOS should be keeping the time between them. I can experiment by changing the task priorities.

davinbirdi, 2022-02-27:
Will test FreeRTOS delayUntil() vs. delay() for periodic tasks

davinbirdi, 2022-02-27:
Tested commenting out the PRINTFs in the ADC Task. This results in the Duty Cycle without any jitter

davinbirdi, 2022-02-27:
Feb27 Notes: I may be writing my TaskDelayUntil Function improperly as it's not having any effect while keeping the printfs. I can try to time the Printfs to see if my delayUntil is too small.

davinbirdi, 2022-03-04:
Changing Task Priority seems to make no difference.
Tested changing Control Task to Priority 1 and keeping Actuator Task at Priority 0.

davinbirdi, 2022-03-04:
Looking into using Segger RTT to get printf data at a rate that does not stop the microcontroller clock as long as the Printf debugging.

davinbirdi, 2022-03-18:
Completed during first week of March using SEGGER RTT. Have made a basic version on main, need to clean it up and will merge.

davinbirdi, 2022-03-20:
Main branch now has SEGGER RTT Capabilities so I will close this issue.

closed 2022-03-20 by davinbirdi