UBC-Rocket/WB-DAQ-Firmware #14: Write Pressure Transducer to SD Card

davinbirdi, 2022-01-16:
For running cold-flow, need to be able to read pressure from tank. We want to write this to an SD Card so we can analyze/see the data after the cold-flow.

davinbirdi, 2022-01-16:
@misprit7 can you update me on if this is currently working for this board? The FC has some code but I don't know of it working on the DAQ

misprit7, 2022-01-18:
Good question. The code right now is written for the flight computer, and it should be functional. We are going to have to find some way of porting it to the DAQ board. Given this is going to a semi-regular occurrence in general (e.g. we'll need to do the same thing for the CAN bus code), we should probably sit down and figure out a way to port code written for the fc/SIM to the DAQ board in at least a semi-modular way.

davinbirdi, 2022-03-20:
Currently Can log to a file using RTT Viewer, we eventually will need the SD Card reading but for now this is on pause.