UBC-Rocket/WB-DAQ-Firmware #24: Main Control Task

davinbirdi, 2022-04-10:
The DAQ board needs a way to interpret CAN commands coming from the Flight Computer to activate certain features, i.e. DPR only turns on when FC tells it to.

Discussed on April 10, 2022 Sunday Morning Meeting, a way to do this is to have one main task that is waiting to receive commands from either CAN or RTT that interprets them and activates the specific feature.

We would like it to be modular so it works the same if you are using RTT or receiving the CAN messages from the FC.

davinbirdi, 2022-07-22:
FreeRTOS Documentation:
https://www.freertos.org/a00018.html

Shawn Hymel Example:
https://www.digikey.ca/en/maker/projects/introduction-to-rtos-solution-to-part-5-freertos-queue-example/72d2b361f7b94e0691d947c7c29a03c9

Use FreeRTOS Queue to manage incoming all incoming messages (whether CAN or RTT)

davinbirdi, 2022-10-07:
Discussed circular buffer/queue with Eric a while back,

@Amarantheum we can follow up with this if you are interested

Amarantheum, 2022-10-09:
Sure. I'd be interested in looking into this.