UBC-Rocket/WB-DAQ-Firmware #30: SPI

Esouder, 2022-09-18:
Not sure of the state of SPI on the DAQ board. We need it to talk to the switches, ADCs, and TCs, Davin believes. This is necessary to use the new board for RCS.

The drivers are already implemented on the FC.

SumairShergill, 2022-10-30:
October 23:

Pin 110 should be SCK but was defined as PTC5

Pin 109 should be CS0

Pin 111 = SOUT was ptc 6
Pin 112 = SIN

Pin 106 should be CS1 

Pin 105 = CS2
Pin 104 = CS3 
Pin 95 should be MOSI 

Pin 96 should be MISO

Pin 90 = CS1
Pin 91 = CS0
Pin 92 = SCK1

SumairShergill, 2022-10-30:
October 30:

SPI sending loop works but no oscilloscope readings

davinbirdi, 2022-10-31:
I kept on looking into this and I think we have have only set the pins on the chip and not setting up the peripheral on the board.

ie. we did this step:
[image]

But are potentially having issues here:
[image]

Let's look into this again on Thursday.

davinbirdi, 2022-10-31:
Had to reinstall that specific version of the SDK (v2.8.2) in oder to clear that error message in previous comment. 

Am now able to update the driver files as RTOS-versions. I think what Sumair and I were doing was trying to program the chip to use RTOS functions whereas the actual hardware was setup to operate SPI in its polling configuration.  

[image]

More Visible here: 
[image]

Because I reinstalled my entire SDK, Git is saying that I modified a bunch of driver files and now I'm unable to compile...

davinbirdi, 2022-11-07:
Today we found that HS Switches require the RSTB pin to be held HIGH in order to use the SPI module on the chip. The DAQrevB Board is missing this electrical connection and requires a jumper in order to continue testing.

So far we have SPI outputting any byte stream we input working well with RTOS coming out of our chip.

Despite this setback of the hardware needing some extra work, we can try to communicate with the other devices on the board (i.e. the ADCs)