UBC-Rocket/WB-DAQ-Firmware #31: Validate that the Control Task Doesn't interfere with PWM Signals

davinbirdi, 2022-10-07:


davinbirdi, 2022-10-07:
@Amarantheum I mentioned this today and we try to figure out a systematic way of testing the timing.

davinbirdi, 2022-10-07:
https://www.freertos.org/vtaskdelayuntil.html

This is the FreeRTOS API Reference/Documentation, I save this as a bookmark as I come back to it a lot: https://www.freertos.org/a00106.html

Esouder, 2022-10-07:
This should ultimately not be an issue though since once we have SPI working we can make the high-side switches handle all the PWM, right?

Amarantheum, 2022-10-09:
I was thinking that maybe a way we could test this is by writing to a buffer every time a task executes. Assuming getting a timestamp takes a small number of cpu cycles, writing to a buffer and updating a pointer into the buffer to write the next task completed and time stamp should take only a few cpu cycles. This is because writing is not like read where the cpu has to wait for slower cache performance for the value to be in the cpu register and the C compiler should optimize a counter in a while loop to just use a register and not read/write to a location in memory every time (unless you use the volatile keyword).