UBC-Rocket/WB-DAQ-Firmware #35: Thermocouple I2C

Esouder, 2022-10-30:


br-zhou, 2023-01-16:
Found last working commit for I2C: 9d683f480690ab93fbb5d05b615fba2df73f52eb
Found first compilable commit where I2C is not working: 0b09ca612769c21db0a07016b56a051a13c445c6

Since no I2C Fireware code was chanced between these two commits, we are investigating the changes in the Peripheral Drivers between the two commits, which appears to have deleted relevant I2C code.

br-zhou, 2023-01-29:
Made I2C work on latest git commit. 
Issue was in the WB-DAQ-Firmware.mex file, in which the I2C peripherals were deleted.

closed 2023-01-29 by br-zhou

br-zhou, 2023-01-29:
NOTE: The TC at address 0b1100000 on I2C1 for the **old** DAQ Board is not working like the rest of the TCs. The chip seems to update the register at 1/4th the frequency of the other TCs.